intel/zephyr.js #1858: [linux] Object cannot be print on Linux

qiaojingx, 2018-03-12:
### Description

This is a regression issue. Object cannot be print on Linux, it causes `Segmentation fault`.
 
### Test Code

[Buffer.js](https://github.com/01org/zephyr.js/blob/master/samples/Buffer.js);
[BufferMemory.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/BufferMemory.js);
[Console.js](https://github.com/01org/zephyr.js/blob/master/samples/tests/Console.js);
[test-console-manual.js](https://github.com/01org/zephyr.js/blob/master/tests/test-console-manual.js); 

### Steps to Reproduction

`1. make BOARD=linux`
`2. ./outdir/linux/release/jslinux samples/Buffer.js`

### Actual Result

 `Segmentation fault` occurs.
[image]

### Expected Result

No error.
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result
---|---|---|---|---
master | 9db8f2b | Linux | Mar.12, 2018 | Fail
 
### Additional Information

xiaoyu3x, 2018-03-12:
This issue is reproduced on STM32F407G-DISC1 and Olimex STM32-E407 with commit [9db8f2b](https://github.com/intel/zephyr.js/commit/9db8f2b5d523b4a09bedf36f6f350ef91b6bae3a)

closed 2018-03-20 by grgustaf

qiaojingx, 2018-03-21:
Verified with commit 3d4a43c in master. This issue is fixed.

PR #1861 (2018-03-20): [console] Fixed console print segfault on arrays
The ZVAL_MUTABLE string was not initialized to a jerry_value_t
when the function returns, the auto-release mechanism will
will segfault.

Fixes #1858
commit 3d4a43c0 (2018-03-20): [console] Fixed console print segfault on arrays (#1861)

The ZVAL_MUTABLE string was not initialized to a jerry_value_t
when the function returns, the auto-release mechanism will
will segfault.

Fixes #1858