intel/zephyr.js #1868: [websocket] Fail to connect with WebsocketServer after zephyr updated to 1.11

xiaoyu3x, 2018-04-12:
### Description

 This is a regression issue, after zephyr updated to 1.11. NodeWebSocketClient cannot connect to WebsocketServer.

### Test Code

[WebSocketServer.js](https://github.com/intel/zephyr.js/blob/master/samples/websockets/WebSocketServer.js);
[NodeWebSocketClient.js](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js);
[WebSocketServer4.js](https://github.com/intel/zephyr.js/blob/master/samples/websockets/WebSocketServer4.js); 
[NodeWebSocketClient4.js](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient4.js); 
[test-ws4-server-dhcp.js](https://github.com/intel/zephyr.js/blob/master/tests/test-ws4-server-dhcp.js)
[test-ws4-client-dhcp.js](https://github.com/intel/zephyr.js/blob/master/tests/tools/test-ws4-client-dhcp.js)
[test-ws4-server.js](https://github.com/intel/zephyr.js/blob/master/tests/test-ws4-server.js)
[test-ws4-client.js](https://github.com/intel/zephyr.js/blob/master/tests/tools/test-ws4-client.js)
[test-ws6-server.js](https://github.com/intel/zephyr.js/blob/master/tests/test-ws6-server.js)
[test-ws6-client.js](https://github.com/intel/zephyr.js/blob/master/tests/tools/test-ws6-client.js)

### Steps to Reproduction

1. `make JS=samples/websockets/WebSocketServer.js ROM=256`
2. `make dfu`
3. set up BLE connection
`echo "connect FF:EE:DD:CC:BB:AA 2" > /sys/kernel/debug/bluetooth/6lowpan_control`
`ip -6 route add 2001:db8::/64 dev bt0`
`ip -6 addr add 2001:db8::2 dev bt0`
4. `Node samples/websockets/NodeWebSocketClient.js`

### Actual Result

[image]

### Expected Result

[image]
 
### Test Builds

Branch | Commit Id | Target Device | Test Date | Result | Comment
---|---|---|---|---|---
master | 3f8eb85| Arduino 101 | Apr 11, 2017 | Pass |
master | 3f8eb85| FRDM-K64F | Apr 11, 2017 | Pass(IPv4) | IPv6 failed by #1842 
master | 3f8eb85| Linux | Apr 11, 2017 | Pass |
master | d5d5540 | Arduino 101 | Apr 11, 2017 | Fail |
master | d5d5540 | FRDM-K64F | Apr 11, 2017 | Fail |
master | d5d5540 | Linux | Apr 11, 2017 | Fail |

### Additional Information

qiaojingx, 2018-05-18:
Verified with commit 2273038 in master. Result is as below:
On my Linux(Ubuntu 16.04): This issue is still reproduced. All websocket server can't be connected with error "Unexpected server response:{}". But it works fine with zjs-0.5.

On another Linux(Ubuntu 16.04): This issue is not reproduced. Websocket server can be connected.
I am not sure whether it is caused by test environment or network connection issue.
So close it for now, and I will create another issue if new problem occurred.

closed 2018-05-18 by xiaoyu3x

haoxli, 2018-05-24:
@grgustaf  I investigate that this issue occurs by node ws module required by [NodeWebSocketClient.js](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js). There are changes in protocol implementation between v5.0.0 and v5.1.0 of [ws/lib/websocket.js](https://github.com/websockets/ws/blob/master/lib/websocket.js):
- websocket.js (5.0.0)
```
if (!protocols) {
     protocols = [];
} else if (typeof protocols === 'string') {
     protocols = [protocols];
} else if (!Array.isArray(protocols)) {
    options = protocols;
     protocols = [];
}
```
- websocket.js (5.1.0)
```
if (Array.isArray(protocols)) {
    protocols = protocols.join(', ');
} else if (typeof protocols === 'object' && protocols !== null) {
    options = protocols;
    protocols = undefined;
}
```
When we use  ws  5.1.0 above, the protocols `["first", "my_ws_protocol", "last"]` in `NodeWebSocketClient.js` will be joined with `', '`,  [websocket server](https://github.com/intel/zephyr.js/blob/master/samples/websockets/WebSocketServer.js) receive the protocols with a white space before each protocol expect the first one:
```
var wss = new WebSocket.Server({
    port: 8080,
    acceptHandler: function(protos) {
        for (var i = 0; i < protos.length; i++) {
           console.log('protos:' + protos[i])
            if (protos[i] == "my_ws_protocol") {
                return protos[i];
            }
        }
        return false;
    }
});
```
output: 
[image]
No protocol match with `my_ws_protocol`,  there is error thrown with 'unexpected server response'.

In Node ws module, the [server](https://github.com/websockets/ws/blob/master/lib/websocket-server.js)  trim the received protocols to align with client.
```
if (protocol) {
     protocol = protocol.trim().split(/ *, */);
```
We are not sure the end-user always install latest ws module, so could you do the same trim for protocols in zjs code? or expose trim() API to handle it in JavaScript code, now there is no such kind of method for String.