intel/zephyr.js #1873: Fail to build STM32F407G-DISC1 and Olimex STM32-E407 images

xiaoyu3x, 2018-04-19:
## Issue description
Can't build STM32F407G-DISC1 and Olimex STM32-E407 images

## Steps to reproduce
Build  STM32F407G-DISC1 or Olimex STM32-E407  image.
`make BOARD=stm32f4_disco ide` or `make BOARD=olimex_stm32_e407 ide`

## Actual Result

‘ST_STM32_OTGFS_50000000_* ’ undeclared here (not in a function)
[image]

## Expected Result
Build image successfully.

## Test Build
Branch| Commit id | Target Device |Test Date | Result 
---| ---| ---| ---|--- 
master | [82662d6](https://github.com/intel/zephyr.js/commit/82662d65bb71d3088db9368d390872f9ccf82ede) | STM32F407G-DISC1/Olimex STM32-E407 | Apr 19, 2018 | Fail 
master | [3d4a43c](https://github.com/intel/zephyr.js/commit/3d4a43c0785ee5d0f972d5bfb6020342ff86210c) | STM32F407G-DISC1/Olimex STM32-E407  | Apr 19, 2018 | Pass

mandarcthorat, 2018-05-08:
I'm also facing this issue.

grgustaf, 2018-05-09:
@mandarcthorat which board were you trying to use?

mandarcthorat, 2018-05-11:
@grgustaf I was trying it on stm32f4 discovery board