intel/zephyr.js #1899: Upgrading to Zephyr 1.14 LTS

pfalcon, 2019-04-18:
I see that there was no commits since summer last year, but we (Linaro) would still be interested to maintain Zephyr.js, at least as a non-trivial application example for Zephyr. (Plus, scripting language adoption may be slow, but it's clear that many people are interested in that, and numbers only grow (slowly, again)).

So, we'd like to update Zephyr.js to be buildable against recently released Zephyr 1.14 LTS, and keep running it in our CI (https://ci.linaro.org/view/lite-iot-ci/job/lite-aeolus-zephyr.js/). I already made a few changes in https://github.com/pfalcon/zephyr.js/tree/zephyr-1.14 branch, but so a few ports still have build issues. I intend to fix them, and then post the patchset upstream.

This is a notice to let interested parties be in loop and avoid duplicated work, and see if the project maintainers are around and would be interested in merging such patches.

Thanks.

pfalcon, 2019-04-18:
@grgustaf, @jimmy-huang, @brianjjones: Ping guys.

brianjjones, 2019-04-18:
Hi @pfalcon !

Currently @grgustaf  is out of the office for the week, but I think we'd be interested in looking at the patches.  Glad to hear someone is still getting some use out of ZJS.

jimmy-huang, 2019-04-19:
@pfalcon Yeah, feel free to submit those patches, always welcome!

pfalcon, 2019-04-26:
Thanks for responses folks, the patchset is posted:

* https://github.com/intel/zephyr.js/pull/1900
* https://github.com/intel/zephyr.js/pull/1901