intel/zephyr.js #1904: connection event listener registration error

emarteca, 2021-03-04:
Hi all, I've been working on a tool to identify instances of events registered to the wrong object in uses of some JavaScript event-driven APIs, as part of a research project.
The tool flagged [line 8](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L8) in samples/websockets/NodeWebSocketClient.js, on the registration of the `connection` event.

The reason I believe this is indicative of an error is as follows (from looking at the [ws API documentation](https://github.com/websockets/ws/blob/master/doc/ws.md#new-websocketserveroptions-callback)). The [`ws` variable](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L7) is a ws.WebSocket.  However, [`connection` is an event on ws.WebSocket.Server](https://github.com/websockets/ws/blob/master/doc/ws.md#event-connection).

The other listeners on `ws` are for [`open`](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L11), [`message`](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L15), [`ping`](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L20), and [`error`](https://github.com/intel/zephyr.js/blob/master/samples/websockets/NodeWebSocketClient.js#L24) all of which are events that do correspond to ws.WebSocket.
So, my guess is that this listener for `connection` should just be removed.

Thanks!

grgustaf, 2021-03-04:
Zephyr.js is actually a separate implementation that is "Node-like" so it can't be directly compared against the Node API. We may have based this on an earlier version of Node or even gotten something wrong in trying to imitate it, but I think in our implementation this is a valid event. Here's a line where it is [emitted](https://github.com/intel/zephyr.js/blob/master/src/zjs_web_sockets.c#L901).

Sounds like a cool project. There might be some way to distinguish our code here from true Node code. Of course, we were trying to make it as similar as we could (but this was targeting very tiny hardware like Arduino).

You might want to figure out a way to limit or prioritize your analysis to projects under active development? Sadly, this one has been inert for several years now. :)

Good luck, and thanks for stopping by! Visit the MFA for me if it's open. :)

emarteca, 2021-03-04:
Thanks for the speedy response! Good point, my tool does not account for manually emitted events so that would probably be the best place to start.

The MFA just re-opened, so I'll go check it out asap :)

closed 2021-03-04 by emarteca